strands-project/scitos_robot #69: Magnetic sensor too insensitive

ToMadoRe, 2015-03-06:
After trying to get the magnetic strip detector working on Werner/Henry today at AAF, @Jailander and I finally came to the conclusion that the sensor is setup a bit too far away or it is too insensitive to magnets.

I dismounted the lower cover from the robot and checked the LED of the sensor. There is no light when I stick the magnetic strip on the ground and put the sensor above, but if I lift the magnetic strip a little bit and bring it closer to the sensor, it goes on. 

@chmartin21, @creuther: Is there a way to fix this?

Jailander, 2015-03-06:
What @ToMadoRe said.

just pinging @marc-hanheide here ;)

marc-hanheide, 2015-03-07:
@chmartin21 @creuther can we make this some priority? Have you had this problem before?

cdondrup, 2015-03-08:
Can the sensor be lowered? Meaning manipulating the mounting to fix it closer to the ground?
Would just be a work-around but might be worth trying if possible.

Can you measure the distance it has to the ground? Maybe we can compare if there are differences and if not its a problem of the sensor itself.

marc-hanheide, 2015-03-08:
I'm going to Vienna on Tuesday and could take our sensor if needed? Not ideal, but it's more urgently needed at aaf.

ToMadoRe, 2015-03-08:
From above, it didn't seem you could lower the sensor easily but I didn't turn the robot to check from the bottom. The robot is at AAF at the moment and it would take me almost 1.5h going there again - if it's just measuring the distance, I would like not to. But maybe @denisehe can measure it somehow.

Let's see what Metralabs tells us to do.

marc-hanheide, 2015-03-09:
The sensor on Henry:

[image]

chmartin21, 2015-03-09:
Unfortunately, the detection quality of the sensors also depends on the environment (e.g. the construction of the floor). You can try to lower the sensor. For that, to the following step:
1. first disconnect the sensor
2. untighten the screw nut below the robot
3. lower the sensor by rotating (but not more than the second screw thread!)
4. fix the screw nut below the robot
5. re-connect the sensor

ToMadoRe, 2015-03-26:
Ok, thanks. It works now but I had to lower it quite a lot.

closed 2015-03-26 by ToMadoRe