strands-project/scitos_robot #70: Power supply for external PCs broken

ToMadoRe, 2015-03-26:
Has anyone else ever had problems with the power supply (g4m11 M3-ATX) of the external PC? One of ours doesn't supply power to the main PCs anymore. (I checked the input voltage to the power supply, which is fine. Plugging in the second power supply to the same PC works.)

As the robot was running for some weeks now, I was just wondering if the power supply is okay for such long-term performance. Therefore, if other power supplies from any of our robots break as well, we might have to consider replacing it by a better one.

cburbridge, 2015-03-26:
Yes we had the same issue. I returned it and got a replacement very fast for free (CarTFT.com). I think the issue is that it is designed for up to 24V but the voltage supplied is not regulated and peaks much higher, shortening the life of it. I noticed that there is a model that accepts higher input voltages, might be worth getting that instead.

If you don't want to return it, try replacing the surface mount fuse on it - that had blown on ours but I didn't try to repair it.

ToMadoRe, 2015-03-26:
Okay, then this is a problem. I have to check with @mzillich where we ordered ours. 

Do you mean this one: http://www.mini-box.com/M3-ATX-HV-DC-DC-ATX-Automotive-Computer-car-PC-Power-Supply ? The voltage input is 6-24V, but the power output is less.

marc-hanheide, 2015-03-26:
That explains, why I couldn't run it one Werner anymore...
But that said, I can confirm that both side PCs have been off for quite a while, so they were not constantly running. 
Anyway, we need this fixed ASAP (pre-deployment!). If you can't get a replacement in time, please let me know. Then we'll have to take Linda's when we come down to Vienna after Easter.

marc-hanheide, 2015-03-30:
Sugested by @cburbridge: http://www.cartft.com/catalog/il/806

Please @mzillich and others should look at this and say if it's suitable.

cburbridge, 2015-03-30:
Would this http://www.cartft.com/catalog/il/806 work instead, providing a
wider input range?

mzillich, 2015-03-30:
Power-wise this looks good. @ToMadoRe can you check if the plugs are ok for our motherboard?

marc-hanheide, 2015-03-30:
OK, can you please make sure we get Henry up and running for the pre-deployment and keep me posted on any expected delays?

Thanks

ToMadoRe, 2015-03-31:
The plug looks okay but the dimension (160x45mm) is much bigger than from the old one. Any ideas how to mount that on the robot?
They also sell these PC-casings (http://www.cartft.com/catalog/gl/55), which would fit the power supply + mainboard + have an additional mount for fan(s). I am afraid though that it won't fit our PCs with the graphics card installed. Can someone help me here please?

mzillich, 2015-03-31:
Did not look at the dimensions in detail, but I am rather sure the graphics card won't fit.

ToMadoRe, 2015-03-31:
I just talked to the distributor and they suggest using the more powerful power supply

M4-ATX 
http://www.cartft.com/catalog/il/903

+ casing
http://www.cartft.com/catalog/il/1066

Any comments on that? It seems definitely more reliable as it has a wider input voltage range + a fan for cooling. I am just worried that the casing won't fit on the robot. Unfortunately, the robot is at AAF so I can't check. But we would have the same issue for the M2-ATX power supply.

mzillich, 2015-03-31:
yup, 250 W should do it. price is also ok, let's take that.
lets take the casing and try, 13 EUR only

ToMadoRe, 2015-03-31:
okay, I have ordered one unit for testing. Let's see if it works.

ToMadoRe, 2015-04-10:
We installed the new power supply and I highly recommend replacing the old one with this one for these reasons:
(1) wider input voltage range (max. 30V instead of 24V), 
(2) it provides much more power (250W instead of 125W), and 
(3) it comes with a fan for cooling and a case that can be easily mounted on the robot

nilsbore, 2015-04-10:
So we should all buy two of the power supply listed above (M4-ATX 6-30V DC/DC (250 Watt)) + two of the case listed above (SFX / ATX Enclosure f. M4-ATX)? That is we have to order the case separately?

cdondrup, 2015-04-10:
http://en.wiktionary.org/wiki/never_change_a_running_system

ToMadoRe, 2015-04-10:
@nilsbore: yes, you would have to order power supply and case seperately. Also note that you need something to connect the two wires from the fan to the connectors on the power supply. Apart from that it is very easy to mount. 

@cdondrup: Well, there have been two broken supplies already even though we haven't really used the graphics card yet. Of course, you can leave it for now; but I think soon or later this might be an issue