strands-project/scitos_robot #71: Upgrading RAM on Central PC

Jailander, 2015-07-17:
Hi we upgraded Linda's RAM to 8GB this week and its working quite well, we think all robots would benefit a lot from doing this, sadly we didn't document the process thoroughly, but I'm posting a picture of the reference for the memories we used, and the one we removed from Linda:

[image]

to fit them in you just need to **carefully** gain access to the PC in your robot and replace as you would in any other PC.

cdondrup, 2015-07-20:
Just as a short descriptions:

1. Unplug all the cables from the PC and remember where they go, some fit in several slots.
1. Unscrew the screws fixing the PC to the frame
1. Loosen the screws holding the box above the PC with the external cards like the EBC control and gyro in it.
1. Lift the box on top of the PC slightly
1. Pull PC out to the front, lifting it above the cables sticking out of the robot base
1. Open back of PC (unscrewing the plate holding the fan)
1. Replace RAM
1. Put everything back together
1. Watch the large memory bar in htop

Please make sure I didn't forget anything @Jailander

Jailander, 2015-07-20:
nope, you didn't miss one word, thank you :+1:

cburbridge, 2015-08-10:
Thanks @cdondrup/@jailander  these are great instructions and now Bob has 8Gb too :-)
Just a small warning for anyone left to do this: the wires coming out of the back are quite easily broken when sliding the PC back in, especially the emergency sensors wire. When I put the PC back in I got that cable trapped behind it and ripped off the black plug...oops.