strands-project/scitos_robot #73: Replacing Battery in Linda?

Jailander, 2016-04-11:
One of the battery cells in Linda is apparently damaged according to analysis ran by metralabs over this [mira tape](labe.felk.cvut.cz/~tkrajnik/miratapebattery.tape). @creuther what is the recommended course of actions here?

creuther, 2016-04-11:
Unfortunately, the battery cell manufacturer specifically states that the battery cells have to be considered damaged if they drop below 2V. As this happened a couple of times, it is certainly safe to say that the battery needs to be replaced. I'll get you a quote for it.

Any heat issues that you may have with Linda might be exacerbated by the defective battery cell(s) as well, as they can produce a considerable amount of heat once they are defective.

Jailander, 2016-04-12:
Yes we thought so... thank you for getting us the quotation.

About the heat issues we have been having them for a while now, I'm sure we can run a comparison on the inside temperature whilst charging before and after we change the batteries if that is useful for you.

creuther, 2016-04-12:
If it is not too much of an effort then it would certainly help to exclude / identify that as one influencing factor. I am waiting to hear back from my boss and will let you know once I got the quote.

creuther, 2016-04-15:
I finally got a quote. So my bosses tell me that a new battery is 950 Euro (net), not including shipping. As lithium batteries are unfortunately classified as dangerous goods (and also they are heavy), the shipping is usually not too cheap either.
So knowing that, please let me know what you want to do. If you want to get the replacement battery then I can have a formal quote sent to you and we can prepare everything in the mean time.

cdondrup, 2016-04-15:
Since @marc-hanheide is sitting on our pot of gold, it would be up to him to decide. In general we seem to get a decent battery life out of Linda but it might be useful for future long-term tests. As we have seen on Werner it will most likely not fix the heat problem though...

@creuther how crucial is it from a technical point of view to replace the battery? Is there any danger in running the robot as it is now?

marc-hanheide, 2016-04-15:
Exactly, I need to understand better what we would gain. Is there a real problem here, or is it just we would see getting 30 minutes more of runtime out of it? Also, I understood one cell was broken? So, is this for the exchange of all cell or just one?
Then, finally, it will be easier for us to buy this replacement while STRANDS is still running, so if we believe we'll need a new battery, then we ought to be buying it soon.

marc-hanheide, 2016-04-15:
Also @tduckett should have a look at this.

cdondrup, 2016-04-15:
I just remembered something, @creuther where does the broken cell diagnosis come from? If it is from the persistent error logs, it might be that it shows the broken pin we had in the battery connector at some point. The voltages themselves do look quite balanced:

```
header: 
  seq: 5719196
  stamp: 
    secs: 1460711806
    nsecs: 748912000
  frame_id: ''
voltage: 28.8010005951
current: 0.00900000054389
lifePercent: 100
lifeTime: -1
charging: True
powerSupplyPresent: True
cellVoltage: [3.6010000705718994, 3.6010000705718994, 3.5980002880096436, 3.6010000705718994, 3.6010000705718994, 3.5980002880096436, 3.6010000705718994, 3.6000001430511475]
```

So if it is from the persistent error log we might have "fixed" that already.

Jailander, 2016-04-15:
@cdondrup I think that diagnosis came from the tape I recorded a couple of weeks ago. The broken pin was fixed by then.

To contribute in the discussion, currently we get a pretty decent Battery life in Linda (~4.5 hours with everything running) I wouldn't spend >1000€ on a new Battery. Unless I'm wrong (please correct me if so @creuther ) having a broken cell can only damage the battery further (and it will have a shorter life cycle) but no other component on the robot.  So, if it makes sense to buy a new battery whilst the project is running I would buy the new battery towards the end of the project if there is extra money and that way we'll guarantee a longer battery life on the robot.

creuther, 2016-04-15:
@marc-hanheide There might be a real problem, as the battery cell manufacturer does not guarantee defined operation of the cells once they've been depleted below 2V (apparently even once) by letting the robot discharge fully without turning it off. The persistent error file stored by the charger showed that this happened a few times on Linda, with cell 1 always being the most depleted one (one time even down to 0V). This led us to conclude that this cell has to be considered as damaged.

Now, as @cdondrup has shown, it doesn't really mean that the cell won't charge fully anymore and thus battery life would be reduced permanently. Rather, it changes the way the cell reacts so it might lose or even gain voltage quicker as other cells. This is a problem as the cells have to have an equal voltage, so the charger will go into the so called "balancing mode" where it tries to restore balanced voltages among the battery cells. This leads to severely restricted charging currents when the cell voltages diverge, which could happen more often now due to the defective battery cell.

@Jailander and all, aside from the behaviour described above, a defective battery cell could also pose a fire hazard as the chemical reaction inside is permanently altered, resulting in the cell giving off excessive heat. No robot has caught fire because of this yet but apparently one robot in China got sort of close to that.

Replacing just a single cell is quite complex mechanically, and as the battery cells age together, it will introduce a different kind of discrepancy into the internal battery chemistry. So we would recommend replacing the entire battery.

If you want to do an additional test, you can create a tape / rosbag of the charger status during an entire charging and discharging cycle, and then plot the cell voltages and look for discrepancies. The tape recorded by @Jailander already showed cell 1's voltage lagging behind though, so I suspect it would look somewhat similar.

creuther, 2016-11-14:
@marc-hanheide @Jailander How has Linda been getting on? Is there still the wish or need to replace the battery?

marc-hanheide, 2016-11-14:
yes, we indeed plan to have the battery replaced, and we also sent some parts to MetraLabs.

creuther, 2016-11-15:
Ok, shall I initiate the process on the MetraLabs end and get you a quotation, and maybe already get a battery ready (depending on how you would want to proceed)?

marc-hanheide, 2016-11-15:
@Jailander can you comment? I know you took some steps already to potentially "freshen-up" Linda. @creuther can you get me a quote?

Jailander, 2016-11-15:
Hi yes a quote wouldn't be bad, when we said about rejuvenating Linda we had a new battery and themain PC in mind but since https://github.com/strands-project/scitos_robot/issues/76 not sure if we need a new pc anyway

creuther, 2016-11-17:
@marc-hanheide Getting the quote for you and will send you it by email. I'll try to get an update on #76 as well.