strands-project/scitos_robot #74: slow cameras on betty

bfalacerda, 2016-05-03:
Our setup in betty is the head cam on the right pc, and the chest cam on the main one. I'm having issues with the cam topics frequency which might be problematic.

The chest works at normal rate (30) for depth, but it's intermittent for rgb. Note that for this one there's no network issues as all is in the main pc:

```
strands@betty:~$ rostopic hz /chest_xtion/depth/image_raw                                                                                              
subscribed to [/chest_xtion/depth/image_raw]
average rate: 29.922
        min: 0.030s max: 0.038s std dev: 0.00163s window: 29
average rate: 29.976
        min: 0.030s max: 0.038s std dev: 0.00140s window: 59
average rate: 29.960
        min: 0.030s max: 0.038s std dev: 0.00138s window: 89
average rate: 29.964
        min: 0.030s max: 0.038s std dev: 0.00141s window: 118
average rate: 29.957
        min: 0.030s max: 0.038s std dev: 0.00137s window: 149
average rate: 29.964
```

```
strands@betty:~$ rostopic hz /chest_xtion/rgb/image_raw                                                                                                
subscribed to [/chest_xtion/rgb/image_raw]
average rate: 14.167
        min: 0.028s max: 0.134s std dev: 0.03701s window: 9
average rate: 14.957
        min: 0.028s max: 0.239s std dev: 0.05528s window: 27
average rate: 15.843
        min: 0.028s max: 0.239s std dev: 0.05137s window: 44
average rate: 16.407
        min: 0.027s max: 0.239s std dev: 0.04575s window: 62
average rate: 19.048
        min: 0.027s max: 0.239s std dev: 0.03974s window: 91
average rate: 19.208
        min: 0.027s max: 0.239s std dev: 0.03873s window: 102
no new messages
no new messages
no new messages
no new messages
no new messages
no new messages
no new messages
no new messages
average rate: 7.368
        min: 0.027s max: 8.585s std dev: 0.84163s window: 103
no new messages
```
The head seems to have some issues due to the network not being fast enough. I don't know if these rates are ok:

```
strands@betty:~$ rostopic hz /head_xtion/depth/image_raw
subscribed to [/head_xtion/depth/image_raw]
average rate: 19.010
        min: 0.051s max: 0.053s std dev: 0.00050s window: 18
average rate: 19.013
        min: 0.051s max: 0.054s std dev: 0.00060s window: 37
^Caverage rate: 19.010
        min: 0.051s max: 0.054s std dev: 0.00060s window: 54
```

```
strands@betty:~$ rostopic hz /head_xtion/rgb/image_raw
subscribed to [/head_xtion/rgb/image_raw]
average rate: 12.674
        min: 0.076s max: 0.082s std dev: 0.00144s window: 11
average rate: 12.669
        min: 0.076s max: 0.082s std dev: 0.00106s window: 24
average rate: 12.672
        min: 0.076s max: 0.082s std dev: 0.00087s window: 37
average rate: 12.673
        min: 0.076s max: 0.082s std dev: 0.00077s window: 49
^Caverage rate: 12.674
        min: 0.076s max: 0.082s std dev: 0.00074s window: 59
```

Can someone that know how cameras are supposed to work tell me something about this? Are the head rates something to worry about? What's going on with chest rgb publishing?

RaresAmbrus, 2016-05-03:
Is your camera plugged into a USB2 or USB3 port? Maybe try switching things around, that always helps (eventually) with openni. Btw. if the framerate drops for the depth on the head cam you might have an issue with the sweeps ....

bfalacerda, 2016-05-05:
chest seems to be fineafter changing port. we still havent figured out why the network is slow and we're not getting the full head cam rate. @RaresAmbrus you think these values I posted above are problematic?

RaresAmbrus, 2016-05-09:
@bfalacerda ~20 Hz for the head camera should be fine.

closed 2016-11-26 by bfalacerda