strands-project/scitos_robot #76: FTDI disconnecting

gestom, 2016-10-21:
Lately, LINDA suffers from quite serious HW problems that are probably related to the serial connection between the PC, CAN bus and the laser scanner. Basically, we get this kernel message:

[33075.354321] usb 2-1.8: USB disconnect, device number 6
[33075.354696] ftdi_sio ttyUSB0: FTDI USB Serial Device converter now disconnected from ttyUSB0
[33075.354725] ftdi_sio 2-1.8:1.0: device disconnected
[33075.355070] ftdi_sio ttyUSB1: FTDI USB Serial Device converter now disconnected from ttyUSB1
[33075.355091] ftdi_sio 2-1.8:1.1: device disconnected
[33075.355186] ftdi_sio ttyUSB2: error from flowcontrol urb
[33075.355261] ftdi_sio ttyUSB2: FTDI USB Serial Device converter now disconnected from ttyUSB2
[33075.355274] ftdi_sio 2-1.8:1.2: device disconnected
[33075.355410] ftdi_sio ttyUSB3: FTDI USB Serial Device converter now disconnected from ttyUSB3
[33075.355428] ftdi_sio 2-1.8:1.3: device disconnected

and we have to completely restart the robot. This happens several times a day.

creuther, 2016-10-26:
I'll make some inquiries whether we've seen anything like this before, and failing that whether there are any diagnostics you can run to help us identify the issue.

creuther, 2016-10-26:
There isn't a lot of room for diagnostics with this specific problem unfortunately, it could / should be a problem on the connection between the mainboard and the MCU addon - could be a hardware fault (defective contact), electrical interference, etc. 

Our suggestion is us sending you a new cable for this connection. @gestom can you let me know the address and c/o of where and whom to send it to?

gestom, 2016-10-26:
Thanks, please send the cable to:

Tomas Krajnik,
Lincoln Centre for Autonomous Systems, 
University of Lincoln, Brayford Pool, Lincoln, LN6 7TS
United Kingdom

creuther, 2016-10-26:
The cable is being sent by [DHL tracked & signed](https://nolp.dhl.de/nextt-online-public/set_identcodes.do?idc=RL742528265DE&extendedSearch=true&lang=en). It might take a bit for the tracking link to become active. Please let me know when you receive it and whether you need help installing it.

Jailander, 2016-11-04:
@creuther can you confirm which the cable we ought to change, we only see this candidate but the cable looks different
[image]

creuther, 2016-11-04:
I've asked Rüdiger and will let you know asap, as unfortunately I don't know either....

Jailander, 2016-11-04:
Hi, @creuther this morning Linda was not detecting the usbs after restart, so we replaced the cable for the one you sent us, but still nothing the pc is unable to see those usb devices. 
What do you think we should do now?

creuther, 2016-11-04:
If you can't get USB devices to work under any circumstances, then we unfortunately do have to consider a fault with the PC or related components. 
As unpleasant as it is, I think the best course of action would be to send us the part of the electronics boards that include the PC and MCU. You need to carefully remove the cables and then loosen the screws at the sides and then you should be able to take it out. Can you send it to:

MetraLabs GmbH
z.H. Rüdiger Scheidig
Am Vogelherd 22
D-98693 Ilmenau
Germany

I wish I could give you better news or a quicker resolution but with the robot driving around for 3 years now, I'm told a hardware fault is not as unlikely anymore...

Jailander, 2016-11-04:
Thank you christian, yes she is getting old :( I'll check with @marc-hanheide and will probably send it tomorrow thank you.

Jailander, 2017-01-06:
Hi we just installed the replacement components in Linda, we think we connected all the cables correctly, but we still get this problem:

```bash
│····························································
SerialCommS300: unable to open serial port /dev/ttyUSB0                                                                                          │····························································
SerialCommS300: unable to open serial port /dev/ttyUSB0                                                                                          │····························································
SerialCommS300: unable to open serial port /dev/ttyUSB0                                                                                          │····························································
[ WARN] [1483703530.214886300]: Missing head angle publication as MIRA parameter error.                                                          │····························································
SerialCommS300: unable to open serial port /dev/ttyUSB0                                                                                          │····························································
SerialCommS300: unable to open serial port /dev/ttyUSB0                                                                                          │····························································
SerialCommS300: unable to open serial port /dev/ttyUSB0                                                                                          │····························································
SerialCommS300: unable to open serial port /dev/ttyUSB0                                                                                          │····························································
[ WARN] [1483703530.414922829]: Missing head angle publication as MIRA parameter error.                                                          │····························································
SerialCommS300: unable to open serial port /dev/ttyUSB0                                                                                          │····························································
SerialCommS300: unable to open serial port /dev/ttyUSB0    
```

I checked `/dev/tty` but I see no USB devices:

```bash
strands@linda:~$ ls /dev/tty
tty        tty13      tty19      tty24      tty3       tty35      tty40      tty46      tty51      tty57      tty62      ttyS0      ttyS14     ttyS2      ttyS25     ttyS30     ttyS8      
tty0       tty14      tty2       tty25      tty30      tty36      tty41      tty47      tty52      tty58      tty63      ttyS1      ttyS15     ttyS20     ttyS26     ttyS31     ttyS9      
tty1       tty15      tty20      tty26      tty31      tty37      tty42      tty48      tty53      tty59      tty7       ttyS10     ttyS16     ttyS21     ttyS27     ttyS4      
tty10      tty16      tty21      tty27      tty32      tty38      tty43      tty49      tty54      tty6       tty8       ttyS11     ttyS17     ttyS22     ttyS28     ttyS5      
tty11      tty17      tty22      tty28      tty33      tty39      tty44      tty5       tty55      tty60      tty9       ttyS12     ttyS18     ttyS23     ttyS29     ttyS6      
tty12      tty18      tty23      tty29      tty34      tty4       tty45      tty50      tty56      tty61      ttyprintk  ttyS13     ttyS19     ttyS24     ttyS3      ttyS7 
```

and I don't see any CAN device on `lsusb`:
```bash
strands@linda:~$ lsusb
Bus 002 Device 006: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 002 Device 005: ID 046d:c21f Logitech, Inc. F710 Wireless Gamepad [XInput Mode]
Bus 002 Device 004: ID 04e7:0020 Elo TouchSystems Touchscreen Interface (2700)
Bus 002 Device 003: ID 08bb:29b2 Texas Instruments PCM2902 Audio CODEC
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

Do you have any idea what connection we should check?

creuther, 2017-01-06:
Hi Jaime, I got some feedback from Rüdiger for you to try:

Did this happen straight after assembly or was it working for a bit, however short? Can you check the MainControlUnit (MCU) for loose components / elements? Can you unplug all the cables and re-plug them, and maybe also try the second USB pin header on the motherboard? 

If that doesn't help, can you sequentially unplug all  USB devices and check what devices are found with "ls /dev/ttyUSB*" after unplugging each one? Just a note, lsusb will only show the UART/USB converter as "Future Technology Devices International, Ltd FT4232 Quad HS USB/FIFO IC" rather than a true CAN device.

Let me know whether any of this works, if not then it could mean transport damage :-(

Jailander, 2017-01-09:
Hi Christian,

Thank you very much for your help unfortunately we did not manage to get it working (it never did), we removed the board and reconnected all the connectors making sure they were tight enough, we also changed the USB connector to the second USB pin header on the motherboard, and nothing worked we can see all other usb devices but the converter :/ 

What can we do about this? do you have any other suggestion or assuming transport damage is fair enough given this?

Jailander, 2017-01-09:
We can have a skype meeting to check the connections with someone from Metralabs too, just to make sure

Jailander, 2017-01-10:
@creuther :)

creuther, 2017-01-10:
Sorry for not responding earlier, this week is horribly busy for us and especially the hardware department. I'll get back to you today, sorry again.

Jailander, 2017-01-10:
No problem just wanted to know if it had been read

creuther, 2017-01-10:
Unfortunately it seems that we will have to send it back again :-( I'll get some detailed pictures tomorrow that will show you what exactly we need.

creuther, 2017-01-11:
Okay, please send us the entire following component ("electronics case") so we can make sure that everything works well this time and hopefully won't suffer transport damage:

[image]

The red highlighted rectangle indicates which part is the electronics case, and the red highlighted circles indicate the screws that need to be unfastened.

Please send it to:

MetraLabs GmbH
z.H. Rüdiger Scheidig
Am Vogelherd 22
D-98693 Ilmenau
Germany

Sorry for the continuing problems with this...

creuther, 2017-01-11:
@Jailander: Have you sent it already?! We might have found another issue that could very likely be the cause and wouldn't require sending it back. I sincerely hope you haven't sent it yet :-(

I'll update with info on the fix soon.

Jailander, 2017-01-11:
Hi we just stopped the package on the way to the post office (pheww) let us know when you have the info :) thank you!

creuther, 2017-01-11:
Phew indeed! Sorry about that, we found out that a batch of cables we received is faulty - another robot was exhibiting the exact same problems so we feel confident enough that replacing the cable will fix Linda as well. We'll send you the cable asap, I'll update once it is on its way.

creuther, 2017-01-12:
The cable has been sent today, tracking link [here](https://nolp.dhl.de/nextt-online-public/set_identcodes.do?lang=en&idc=RL746076234DE&rfn=&extendedSearch=true). I'll get you a picture as well for which cable to replace, but it should be quite straightforward :)

Jailander, 2017-01-17:
Hi, @creuther the tracking link is not working and we have not received the cable, can you check please if the link is correct, or do you know to whom it was sent?

creuther, 2017-01-17:
Hi, when I posted it I assumed the link wasn't working due to it just having been posted but apparently not. I wrote an email but people have left the office for now... It was sent to this address (as before, from Marc):

University of Lincoln
School of Computer Science
Alison Cooke
MHT Building, 3rd floor
Brayford Pool
Lincoln, LN6 7TS
Phone: +44 1522 88 6966

Jailander, 2017-01-17:
Ok, we will check her mail inbox as she has been out of office, thank you

marc-hanheide, 2017-01-19:
did anyone check if we received the cables yet?

Jailander, 2017-01-19:
yes we called Callum he hasn't received anything for allison

marc-hanheide, 2017-01-19:
👎

r-sc, 2017-01-20:
dear all, this is Ruediger from MetraLabs. Sorry, I provided Christian a not working tracking link. Here is a working one - the Deutsche Post has marked the letter as [delivered](https://www.deutschepost.de/sendung/simpleQueryResult.html?form.sendungsnummer=RL746076234DE&form.einlieferungsdatum_tag=15&form.einlieferungsdatum_monat=1&form.einlieferungsdatum_jahr=2017). I used this stamp: 
[image]
Please let me know, what to do next. New cable to same address? New cable to new address? Search for the last letter?

santosj, 2017-01-20:
@r-sc @creuther We got the cable, it was "lost" in Alison's office. Sorry about this! We are going to replace the cable today and we will let you know if it works.

Many thanks!

r-sc, 2017-01-20:
I keep my fingers crossed :)

Jailander, 2017-01-20:
@r-sc @creuther Linda is finally up and running again! thank you very much

r-sc, 2017-01-20:
also many thanks - great news!

closed 2017-02-10 by marc-hanheide