strands-project/scitos_robot #77: Werner sometimes loses control of one Wheel

Jailander, 2017-02-10:
Hello, Werner sometimes loses control of one Wheel and reconnecting the wires going to the motors is necessary, they are not obviously disconnected but that seems to be the magic fix, @creuther can we do something about this (it was the left wheel both times)

creuther, 2017-02-10:
Hmm, so a "simple" power cycle with the key does not fix the issue? There should really be no difference in reconnecting the wires unless there really is a mechanical fault at work. 

The motors we use have been known to go into an error state on a rare basis, which up to a certain MCU firmware and SCITOS driver version was only fixable by a power cycle (as mentioned above: turning the robot off with the key and waiting for at least a few seconds). The STRANDS robots should all have the required MCU firmware version >= 1.8.9, and according to [this](https://github.com/strands-project/scitos_drivers/blob/indigo-devel/scitos_mira/CMakeLists.txt) file the SCITOS package should be of the required version >= 0.22.7 as well. 

So far so good if this is what happens, but for reasons beyond me, the detection and reset of the motors was disabled by default until SCITOS version 0.26.0. You can enable it yourself by adding these lines in [SCITOS-G5.xml](https://github.com/strands-project/scitos_drivers/blob/indigo-devel/scitos_mira/resources/SCITOS-G5.xml) within the `<MainControlUnit>` tag:

    <MainControlUnit>
    ...
        <MotorResetOnErrorCurrentThreshold>0.03</MotorResetOnErrorCurrentThreshold>
        <MotorResetOnErrorTimeout>500</MotorResetOnErrorTimeout>
    ...
    </MainControlUnit>

This will only help if these are instances of the problem mentioned above. In any case it will be good to add these lines to SCITOS-G5.xml. Please let me know whether it happens again after adding these lines, and also check the MIRA log output at this time whether there is any indication of abnormal behaviour (exceptions, ERROR entries, etc).

marc-hanheide, 2017-02-10:
I created https://github.com/strands-project/scitos_drivers/pull/116

But this should be tested on Linda first, if it has any side effects.

closed 2017-04-05 by hawesie